LaboratorioGluon/Icaro #3: Sensores básicos internos.

LaboratorioGluon, 2025-01-20:
Elegir que sensores vamos a colocar en la tarjeta.

Mínimo:
- Temperatura
- Humedad
- Presión (barómetro)

sergivv, 2025-01-20:
Veo problemas para encontrar sensor de temperatura adecuado.
Los más comunes no tienen suficiente rango negativo. Por el momento el que tiene el mayor rango (negativo) es el ds18b20 que llega hasta los -55ºC pero viendo las sondas de hoy se alcanzaron temperaturas de -62,5ºC.

Buscando en lcsc hay varios que dicen ser hasta -70º pero después ves el datasheet y no pasan de -20º

sergivv, 2025-01-20:
Todo sería probar.
Otra opción sería montar una sonda pt100 que llegan hasta los -200º pero creo que también son carillas.

LaboratorioGluon, 2025-01-21:
Aquí aparecen unos sensores que marcan de -70: https://s.click.aliexpress.com/e/_olsfRwZ
Son PT100, habría que preparar el circuito de adaptación a los sensores.

Prediktor93, 2025-01-24:
En LCSC hay algunos sensores de temperatura PT100 que llegan hasta los -70º, como por ejemplo este:
https://www.lcsc.com/product-detail/Temperature-Sensors_YAGEO-32208548_C17639042.html
He metido este porque con estos rangos de temperatura parece que solo hay dos frabricantes, Mysentech y este, y este es el unico que tenia los datasheets en ingles.

Otra cosa a tener en cuenta, es que el sensor no deberia ir montado sobre la placa, si no con una sonda al exterior, si queremos una lectura real, ya que con el aislante del corcho y el calor de la electronica la lectura va a ser loca. Algo como esto:

[image]

[image]

LaboratorioGluon, 2025-01-24:
El que has pasado @Prediktor93 está out of stock :(

Prediktor93, 2025-01-24:
Pues tocara si no jugarsela con el dataset en chino con los de la otra marca
https://www.lcsc.com/product-detail/Temperature-Sensors_Mysentech-M117P_C5155921.html

david-oltra, 2025-01-24:
Mysentech M1820P
https://www.lcsc.com/product-detail/Temperature-Sensors_Mysentech-M1820P_C3025191.html

Se supone que llega a -70 aunque el datasheet en chino no ayuda...

[image]

sergivv, 2025-01-24:
Lo que mosquea un poco es que miras el datasheet y no aparecen los -70 grados por ningún lado.

david-oltra, 2025-01-24:
> Lo que mosquea un poco es que miras el datasheet y no aparecen los -70 grados por ningún lado.

En la imagen que he puesto pone un error de ±0.5ºc en el rango de -70 a -30. Poco mas he visto que haga referencia a -70.

LaboratorioGluon, 2025-01-26:
Sensores básicos:
Internos:
- BME280 (Temperatura, humedad y presión)
- LM20 (Temperatura PCB)
- GPS

Inter despegado:
 - [Opcional] PT100.

Externos:
- PT100

closed 2025-01-26 by LaboratorioGluon